sinara-hw/DIO_BNC #5: direction control over I2C

gkasprow, 2018-09-25:
@jordens is direction control over I2C implemented?
Tom Walker reports that direction readout works, he can also control it via DIP switch. But when he tries to set it from ARTIQ it does not work.

taw181, 2018-09-25:
To expand on this, the dashboard reads In or Out correctly when setting DIP switch or using TTLInOut.output() or .input(), but I can't output TTL high either via on() or OVR + LVL on dashboard when the switch is set to In. I can output TTL high when the DIP switch is set to out.

hartytp, 2018-09-25:
> To expand on this, the dashboard reads In or Out correctly when setting DIP switch or using TTLInOut.output() or .input(), but I can't output TTL high either via on() or OVR + LVL on dashboard when the switch is set to In. I can output TTL high when the DIP switch is set to out.

Moninj doesn't currently touch the I2C bus that controls the TTL direction.

TTL.{output, input} control the LVDS interface on Kasli, *not* the buffer on the DIO board.

hartytp, 2018-09-25:
See https://github.com/m-labs/artiq/blob/20cddb6a2505ec943592abd515f7373e53331396/artiq/gateware/rtio/phy/ttl_simple.py#L78

jordens, 2018-09-25:
It's not implemented in ARTIQ. The dashboard doesn't know anything about the I2C direction control or DIP switch settings. If you have code, you should clean it up and post it for review.
IIRC I tested the basic functionality of the I2C direction control a while back.

If implemented, this would also hook into https://github.com/m-labs/artiq/issues/823