sinara-hw/DIO_BNC #11: DIO_BNC in 4HP/CPCIS

gkasprow, 2020-06-18:
The DIO_BNC can be easily migrated to 4HP already - I developed a single MCX adapter that can be installed in all EEM modules with BNC connectors. So far we have Sampler, DIO_BNC, and Zotino adapter.However, che CPCIS routes 16 LVDS lanes to every slot. 
To effectively use all lanes, I already developed DIO_MCX, it was produced and works.
I'm collecting ideas about the future of isolated DIO.
Shall we scrap DIO_SMA and DIO_BNC and replace with 16 channel isolated MCX version?
Shall we preserve the idea of isolated quad IOs ? Or, higher granularity is needed?
Or, simply leave existing DIO_SMA, DIO_BNC(MCX) and make an adapter that uses only 8 LVDS lanes.
Yet another idea is to use existing EEM to CPCIS adapter but make a new revision of DIO_SMA and DIO_BNC(MCX) 160mm long. This is my favorite approach because it ensures backward compatibility and no additional RnD for adapter board.
The Kasli and Kasli Zynq have enough slots for 12 x8LVDS slots. With Kasli CPCIS adapter we will route them to 8 CPCIS slots in such a way that some slots would be x16, some x8 and some of them configurable (x8 or x16). With ZUS+ system board, all CPCIS slots would be x16.
Eventually, we will make native CPCIS support to Kasli which will slightly lower its cost and improve robustness. Once we make the ZUS+ system board running with ARTIQ, we will probably scrap Kasli because the ZUS+ FPGA would cost nearly the same as Artix.

jordens, 2020-06-18:
IME the downsides to isolation are significant and the approach is so incomplete that it is rarely helpful. Doing direction on a finer basis would be helpful. Many setups waste 3 channels because they need one input. 
160mm long dio board is ok with me. Better yet skip the phase where we need the cpcis adapter altogether.

gkasprow, 2020-06-18:
OK, so we would replace the 8 channel DIO_SMA and DIO_BNC with 16 channel individually isolated MCX IOs. I'd use an approach already implemented in DIO_MCX with upper mezzanine.
To lower the cost, one can always order a twice cheaper version without the upper mezzanine, with only 8 IOs. That's true, there is no real benefit of having EEM version of it since existing boards can be used...

hartytp, 2020-06-19:
> with 16 channel individually isolated MCX IOs

Echoing what @jordens said above, I think we should remove the isolators from this design. We can always produce a second design with isolators later on if someone really needs them, but I do not think they should be the default (and I don't think it's worth the complexity of having a design that can support both isolated and non-isolated modes).

Isolation is much less of a priority for me than it used to be, primarily due to the success of Kasli, DRTIO, etc. In the old days we used to have one mammoth control system with everything attached, which led to all kinds of grounding nightmares. Now we tend to have more of a "one ground" per Kasli model (the exception there being ADC inputs, which tend to come from far afield, but Sampler takes care of that nicely).

The isolators add complexity (floating supplies) and jitter and can also couple AC noise into the system in unexpected ways if one isn't careful. Better not to have them except in the few situations where they really are nececcary.

gkasprow, 2020-06-19:
@hartytp we already have 16channel MCX DIO up and running :)
CERN is also interested in individually isolated version.

dtcallcock, 2020-06-20:
> That's true, there is no real benefit of having EEM version of it since existing boards can be used...

Wouldn't making the 16x non-isolated MCX board 160mm for now would save having to keep an EEM and CPCIS version around?

> Better not to have them except in the few situations where they really are necessary.

Yeah, I have found having an isolated input for the PMT can be handy.  Kind of wish Kasli had one or two isolated SMA digital inputs on the front panel (perhaps as a little mezzanine board).

gkasprow, 2020-06-20:
@dtcallcock we already have [16xMCX ](https://github.com/sinara-hw/DIO_MCX/wiki) non-isolated 160mm EEM DIO
Together with an adapter, it works with CPCIS and.
If there is enough interest, I can make individually isolated 16channel DIO in 160mm EEM form factor.
Since we already have DIO-BNC and DIO-SMA, the plan is to make CPCIS version only.

hartytp, 2020-06-23:
> Kind of wish Kasli had one or two isolated SMA digital inputs on the front panel (perhaps as a little mezzanine board).

Isn't that "little mezzanine board" the DIO_SMA ? ;)

In all seriousness though, I think it's a feature of Kasli that it doesn't have this kind of thing: there isn't room on the FP for SMAs/D-types so they would have to be mezzanines and everyone wants something slightly different there (a few low-speed DACs, a few ADCs, isolated/non-isolated BNCs/etc). Rather than trying to create something to suite everyone we keep things modular and provide a wide array of EEMs to do the job. If you want something cheaper/different then it's pretty easy for any student to throw something together to serve your purposes...

BTW can't you float your PMT if ground loops are really an issue?

dtcallcock, 2020-06-23:
> Isn't that "little mezzanine board" the DIO_SMA ? ;)

Yes, that's the problem.  It seems like with the current design, most people are wasting 3 inputs.  It sounds even worse under CPCIS where I'd have to replace a 16-ch MCX card with a DIO_SMA I wouldn't otherwise want, just to get one isolated input.  

I think you're saying this isn't a problem though as Sinara isn't trying to 'suite everyone'.  I think this's a reasonable standpoint if the solution makes the DIOs less modular/general in some other way.

> isn't room on the FP for SMAs/D-types so they would have to be mezzanines

It's just one or two SMAs, so actually I think a pigtail from the panel (where there is a bunch of room to the left of the SFP cages) to an MMCX or whatever on the board would actually work here - no mezzanine or risk of people demanding DACs.  

I think the PMT input is a unique and special signal so having it go straight to the FPGA board isn't that unreasonable.  For non-ion-trap applications it can be considered just a general TRIG/SYNC input (probably useful in its own right).

However, I think Kasli has sailed now.  Just wanted to mention that it's a solution to the above.  

> BTW can't you float your PMT if ground loops are really an issue?

In this case there was a HV supply involved so floating wasn't a (sensible) option.  Sure, there are other ways to fix it though.  I just bring it up as a time when having isolation was actually useful.

hartytp, 2020-06-23:
> I think you're saying this isn't a problem though as Sinara isn't trying to 'suite everyone'.
> I think this's a reasonable standpoint if the solution makes the DIOs less modular/general in some other way

My point is that Sinara favors modularity over optimizing for individual use-cases. It's easy to build hardware based on a wishlist from a small number of users and end up with something bloated (c.f. Sayma) that tries to be too much to too many people.

I think that the real success of Kasli is that it's completely minimal; just an FPGA with a tonne of IO broken out to sensible connectors, making minimal assumptions about how it's going to be used. 

You're implying that a couple of isolated IO on FP SMAs is something that would be useful to most users of Kasli. I don't believe that to be the case (but would be happy to be corrected if you have data).

> It's just one or two SMAs, so actually I think a pigtail from the panel (where there is a bunch of 
> room to the left of the SFP cages) to an MMCX or whatever on the board would actually work 
> here - no mezzanine or risk of people demanding DACs.

IIRC there also aren't any spare pins on the FPGA (although I can't remember if that's still true after the changes on v2.0). If there were then I'd be much more in favor of e.g. adding an extra IDC header. People can then easily break them out to whatever they want (e.g. add an I2C DAC instead of the IO if that's more useful to them and they don't want 32 channels of Zotino).  

> I think the PMT input is a unique and special signal so having it go straight to the FPGA board 
> isn't that unreasonable. For non-ion-trap applications it can be considered just a general 
> TRIG/SYNC input (probably useful in its own right).

I don't agree. Many (most?) Kaslis are slaves, not masters and don't connect to a PMT. TRIG/SYNC inputs are pretty rare in ARTIQ systems (I think we scrapped line triggers long ago since they slowed down our data acqusition rate unacceptably).

Again, my concern here is that you're thinking about/optimizing for an individual small-scale experiment rather than the very large range of uses that Sinara serves (admittedly, that would be much easier if the vast majority of users would actually make themselves heard here...).

> In this case there was a HV supply involved so floating wasn't a (sensible) option. Sure, there are 
> other ways to fix it though. I just bring it up as a time when having isolation was actually useful.

Again, IMHO even if we did have an input on Kasli, it shouldn't be isolated. For all the reasons we've discussed elsewhere, that's not completely unproblematic and is a pretty niche use-case. 

If we have spare pins then this feels like something that could be best served by a small IO mezzanine (can hang of the FP) that connects to Kasli from a ribbon cable.

dtcallcock, 2020-06-23:
> Many setups waste 3 channels because they need one input.

This was the original point I was responding to (and is presumably based on data given @jordens is in the crate selling business).  If we don't want to address it on Kasli (and I see you _really_ don't want to do that), do we want to on DIO_BNC (which is after all what this tread is about)?  

How about making the isolation Ch 0-1/2-7 instead of Ch0-3/4-7 and then doing the direction selection for every 2 channels instead of 4?  That would involve pretty minimal changes to the design.

jordens, 2020-06-23:
Imo it would be ok to change the dio boards to support just 1:7 or 0:8 in:out partitioning. I.e. only treat the first channel as special with optional input and optional termination. It would simplify a few things. That also matches the notion of having the first eem signal a clock-capable fpga pair.

dnadlinger, 2020-06-23:
1:7/0:8 only wouldn't work for our use case here – we have equal numbers of TTL inputs and outputs. Urukul (and DRTIO hardware in general) really reduce the need for bare TTL signals compared to traditional dumb-synth-plus-RF-switch-array-type setups.

hartytp, 2020-06-23:
Does having per-channel direction control add to the cost/complexity of the design in a noticeable way?

gkasprow, 2020-06-24:
@hartytp we already have per-channel direction control. The issue is to have per-channel isolation :)
The existing DIO16-MCX has also per-channel direction control but no isolation.

gkasprow, 2020-06-24:
We can add some uFL or MMCX connector + passive protection or buffer to the next Kasli revision. There are still 8 free IO pins. Then a low-cost SMA-UFL pigtail would do the job.

hartytp, 2020-06-24:
aah, sorry, yes.

@dnadlinger correct me if I'm wrong but we're not planning to use any isolated channels are we, so @jordens suggestion wouldn't be an issue for us...

gkasprow, 2020-06-24:
We plan to do design such an isolated board to popularize Sinara HW among the CERN users. They use a lot of isolated IOs for machine protection.

hartytp, 2020-06-24:
> We can add some uFL or MMCX connector + passive protection or buffer to the next Kasli revision. There are still 8 free IO pins. Then a low-cost SMA-UFL pigtail would do the job.

- If we're going to discuss this then we should discuss on Kasli not here
- I'd still prefer to reserve a couple of IO for future use. If we are going to break them out then a header seems to be a better way than an uFL/MMCX. It's still dirt cheap to use a ribbon cable to route to a tiny PCB that mates with the FP

dnadlinger, 2020-06-24:
@hartytp:

Robert wrote

> Imo it would be ok to change the dio boards to support just 1:7 or 0:8 in:out partitioning. 

which would seem to refer to direction.

Switchable isolation would be fine, plus of course there could always be multiple variants in addition to the "standard" DIO-* modules.

hartytp, 2020-06-24:
Indeed, I think there has been some confusion in this thread over exactly what we're talking about.

IIU correctly the situation is that:
- this particular conversation doesn't relate to updating the existing EEMs, but rather to new CPICS DIO EEMs
- there is [already](https://github.com/sinara-hw/DIO_MCX) a 16-channel non-isolated DIO EEM with per-channel direction control. This is what I would consider the standard "standard" DIO choice for CPICs
- this discussion is about making an isolated equivalent

Isolation is something I consider somewhat niche and don't think we'll use heavily, so don't have strong opinions about the optimal partitioning between the grounds.

Obviously users should feel free to create any other variants they want. IME it's rarely worth it however. e.g. it's usually cheaper to be a bit wasteful using modules that are produced in high volumes than to get a custom module made that's exactly tailored to one's specific use-case. Fewer designs produced in higher volumes tends to result in lower overall costs and higher quality products IME

gkasprow, 2020-07-02:
nothing to do here, closing

closed 2021-07-28 by gkasprow