sinara-hw/DIO_BNC #22: IC30 EN pin voltage limit

MorganTL, 2023-08-23:
From the datasheet of IC30, the maximum voltage allowed for the EN pin 6V. But in V1.6.1, the EN pin of IC30 is directly connected to the P12V0 which may damage the IC

[image]
[image]

gkasprow, 2023-08-23:
It's disconnected in 1.6.1 release. But I see the problem in repo. There is a missing DIO-BNC_Supply file in the sources and my version of 1.6.1 has this pin disconnected.The PCB file is OK.
[image]
So the sources are ok and the release is wrong.

gkasprow, 2023-08-23:
I updated the releases

gkasprow, 2023-08-23:
Just lift the pin 5 of the IC30 or cut the trace below the chip

gkasprow, 2023-08-23:
Cti produced 1.6.1 with pin floating :)

MorganTL, 2023-08-24:
I did a xor compare using the previous gerber files with the current release one and found that the track between EN and VIN pin is connected for the previous GTL gerber files. Also there's some changes to the GBL gerber (see attached).

[image]
[image]